atc1441/ESP32_nRF52_SWD #8: Problems compiling the  ESP32_SWD_wifi

madaeon01, 2021-07-05:
Hi,
I followed all the instructions in the guide. For the WifimManager, I manually downloaded "https://github.com/tzapu/WiFiManager/tree/feature_asyncwebserver"
I got the following error when compiling: 

```
"C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::WiFiEvent(arduino_event_id_t, system_event_info_t)':
C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.cpp:3452:17: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     if(event == SYSTEM_EVENT_STA_DISCONNECTED){
                 ^
C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.cpp:3469:20: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
   else if(event == SYSTEM_EVENT_SCAN_DONE){
                    ^
C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::WiFi_autoReconnect()':
C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.cpp:3486:79: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'
       wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2)); // @todo move, needed for async esp32 scannetworks
                                                                               ^
In file included from C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiSTA.h:28:0,
                 from C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFi.h:32,
                 from C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.h:70,
                 from C:\Program Files (x86)\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:
C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiGeneric.h:145:21: note: candidate: wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventCb, arduino_event_id_t)
     wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^
C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiGeneric.h:145:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type {aka std::_Bind<std::_Mem_fn<void (WiFiManager::*)(arduino_event_id_t, system_event_info_t)>(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>)>}' to 'WiFiEventCb {aka void (*)(arduino_event_id_t)}'
C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiGeneric.h:146:21: note: candidate: wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventFuncCb, arduino_event_id_t)
     wifi_event_id_t onEvent(WiFiEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^
C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiGeneric.h:146:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type {aka std::_Bind<std::_Mem_fn<void (WiFiManager::*)(arduino_event_id_t, system_event_info_t)>(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>)>}' to 'WiFiEventFuncCb {aka std::function<void(arduino_event_id_t, arduino_event_info_t)>}'
C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiGeneric.h:147:21: note: candidate: wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^
C:\Users\md\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\src/WiFiGeneric.h:147:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type {aka std::_Bind<std::_Mem_fn<void (WiFiManager::*)(arduino_event_id_t, system_event_info_t)>(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>)>}' to 'WiFiEventSysCb {aka void (*)(arduino_event_t*)}'
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
exit status 1
Errore durante la compilazione per la scheda ESP32 Dev Module."

```

atc1441, 2021-07-06:
Hey, it seems like it is not in connection to the WifiManager but something else,

What ESP32 module have you selected?
What software version of cor and library is installed?

madaeon01, 2021-07-06:
I have the same ESP32 (esp32-wroom-32) board, same pinout.
Arduino version 1,8,15.
Arduino core for Esp32 (tested both 1,0,6 and dev version 2,0,1).
Board selected Esp32 dev module, partition scheme no OTA (1mb/3Mb).
LIbraries downloaded manually from github:
"WiFiManager-feature_asyncwebserver", "AsyncTCP-master",
"ESPAsyncWebServer-master".
Errors happen during compile, before upload.

mrquincle, 2024-05-17:
For me it just worked with

```

cd Arduino/libraries
git clone https://github.com/me-no-dev/AsyncTCP
git clone https://github.com/me-no-dev/ESPAsyncWebServer
git clone https://github.com/paulo-raca/ArduinoBufferedStreams
git clone https://github.com/tzapu/WiFiManager
cd WiFiManager
git checkout feature_asyncwebserver
```

Perhaps you're using a different lib after all?