atc1441/ESP32_nRF52_SWD #9: Swd-clk seems to be powering the chip

Kwant, 2021-07-06:
When trying to run the glitcher on an nrf52832 the target doesn’t fully power cycle. The voltage on dec1 actually increases a bit when IO22 is low. This effect disappears when disconnecting the swd clock line which I think is somehow backpowering the chip.

atc1441, 2021-07-06:
Hey.

Can you try if a digitalWrite CLK LOW before the glitch does help?

I tested it on an nRF52832 without problems

Kwant, 2021-07-06:
Adding that in the line before the power cycle  seems to do the trick.  No luck on a successful glitch yet but what I see on the scope looks correct now.

atc1441, 2021-07-06:
Thanks for the info. Feel free to make a PR with the change

On the nRF52832 it also took longer for me. The area at the voltage drop is still the right place.