atc1441/ESP32_nRF52_SWD #13: Problems compiling the ESP32_SWD_wifi

lwlbeng, 2021-11-20:
In file included from C:\Users\L\Downloads\ESP32_nRF52_SWD-main\ESP32_nRF52_SWD-main\ESP32_SWD_WIFI\ESP32_SWD_WIFI.ino:1:

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

In file included from C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/WiFiManager.h:109,

                 from C:\Users\L\Downloads\ESP32_nRF52_SWD-main\ESP32_nRF52_SWD-main\ESP32_SWD_WIFI\web.cpp:15:

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

Multiple libraries were found for "WiFi.h"

lwlbeng, 2021-11-20:
In file included from C:\Users\L\Downloads\ESP32_nRF52_SWD-main\ESP32_nRF52_SWD-main\ESP32_SWD_WIFI\ESP32_SWD_WIFI.ino:1:

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

In file included from C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/WiFiManager.h:109,

                 from C:\Users\L\Downloads\ESP32_nRF52_SWD-main\ESP32_nRF52_SWD-main\ESP32_SWD_WIFI\web.cpp:15:

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

Multiple libraries were found for "WiFi.h"

lwlbeng, 2021-11-20:
`In file included from C:\Users\L\Downloads\ESP32_nRF52_SWD-main\ESP32_nRF52_SWD-main\ESP32_SWD_WIFI\ESP32_SWD_WIFI.ino:1:

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:7: error: 'wifi_country_t' does not name a type; did you mean 'fsfilcnt_t'?

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

       ^~~~~~~~~~~~~~

       fsfilcnt_t

In file included from C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/WiFiManager.h:109,

                 from C:\Users\L\Downloads\ESP32_nRF52_SWD-main\ESP32_nRF52_SWD-main\ESP32_SWD_WIFI\web.cpp:15:

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:371:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:372:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\L\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver/strings_en.h:373:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};

                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                                              CONFIG_ESP_PHY_MAX_TX_POWER

Multiple libraries were found for "WiFi.h"`

atc1441, 2021-11-20:
Looks like you have multible/ the wrong asynwebserver librarys installed. This project needs a special one mentioned in the readme

Please check that

lwlbeng, 2021-11-20:
Can you give the version of each library??

lwlbeng, 2021-11-22:
The WiFiManager latest code has the following error

`In file included from /Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/WiFiManager.h:109,
                 from /Users/l/winShare/esp32/ESP32_nRF52_SWD-main/ESP32_SWD_WIFI/web.cpp:15:
/Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/strings_en.h:371:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/strings_en.h:371:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'
 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              CONFIG_ESP_PHY_MAX_TX_POWER
/Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/strings_en.h:372:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/strings_en.h:372:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'
 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              CONFIG_ESP_PHY_MAX_TX_POWER
/Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/strings_en.h:373:46: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' was not declared in this scope
 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/l/Documents/Arduino/libraries/WiFiManager-feature_asyncwebserver/strings_en.h:373:46: note: suggested alternative: 'CONFIG_ESP_PHY_MAX_TX_POWER'
 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              CONFIG_ESP_PHY_MAX_TX_POWER`

lwlbeng, 2021-11-22:
WiFiManager version 0.16 has the following error
`In file included from /Users/l/winShare/esp32/ESP32_nRF52_SWD-main/ESP32_SWD_WIFI/web.cpp:15:
/Users/l/Documents/Arduino/libraries/WiFiManager-78f00ffe0a579609a9da922e1fa0df9e81b80296/WiFiManager.h:16:10: fatal error: ESP8266WiFi.h: No such file or directory
 #include <ESP8266WiFi.h>
          ^~~~~~~~~~~~~~~
compilation terminated.`

vgmjrp, 2021-11-29:
any updates? I'm having exactly the same issues. I strictly used the linked libs (clean sheet Arduino IDE) and even tried different commits of the asyncwebserver-branch. The issue were also reproduced on a portable Arduino IDE.

atc1441, 2021-11-29:
Could you please try it with PlatformIO as well ?

vgmjrp, 2021-11-29:
Seems to be an issue with the ESP32 core (2.x.x) and the  asyncwebserver-branch. ->https://github.com/espressif/arduino-esp32/issues/5821

My workaround was to simply use ESP32 Ver. 1.0.6. Perhaps you can add this information to README.md

atc1441, 2021-11-29:
Added it to the readme just now

ANTI443, 2021-12-07:
Just to add, same issue... v1.0.6 worked for me also!

ArduinoOTA : by Juraj Andrassy Version 1.0.6

Thanks!