atc1441/ESP32_nRF52_SWD #19: Glitching through DEC4 instead of DEC1?

DownrightNifty, 2023-01-13:
Firstly, thanks a ton for putting this together! I'm new to voltage glitching and I'm trying to glitch an nRF52832-CIAA but it's in a weird package ([MBN52832](https://www.murata.com/products/productdata/8813650903070/mbn52832.pdf)) made by a company called Murata which doesn't expose the DEC1 pin. Do you think glitching would be possible through the main power supply (DEC4) instead of through DEC1 (CPU power) if I removed the capacitors?

epoide, 2023-01-17:
If DEC1 is not really exposed through a capacitor what you can do is desolder the nRF chip and provide voltage to it and you can access directly to the VDD_CORE . Since this is only to enable jtag and extract the flash , this approach could work.

DownrightNifty, 2023-01-20:
Looking at the datasheet for the MBN52832 (see page 10 for pinouts): https://www.murata.com/products/productdata/8813650903070/mbn52832.pdf

It appears that the only available pins are VDD and and DEC4.

epoide, 2023-01-20:
I meant disolder the nrf BGA from the murata module.

DownrightNifty, 2023-01-20:
Ah, I didn't know that was possible! Not sure how to go about doing that.

c0d3z3r0, 2023-02-22:
Check out #10 ;)