atc1441/ESP32_nRF52_SWD #22: Flashing bootloader

bbuster2, 2023-02-20:
I am trying to flash the bootloader on a brand new chip (Ebyte E73 nRF52840).
Checkin if the chip is locked gives me the following result: Ok: the nRF is unlocked

I downloaded a few bootloaders including the SEEED USB bootloader: Seeed_XIAO_nRF52840_Sense_bootloader-0.6.1_s140_7.3.0.hex

I can flash the chip and i see after a few moments the chip is written

> ....
14:55:22.959 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:22.992 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:22.992 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.025 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.025 -> 000 Write Register: 0x4001e504 : 0x00000001
14:55:23.025 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.057 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.057 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.090 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.090 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.121 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.121 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.121 -> Done flashing file, it took 19015ms speed: 28.0205kbs

Is this the correct result?
I used the SEEED bootloader because i try to flash my own software via Arduino to the nRF52840. I connected a usb cable to the VBUS (5v), GND, D+ and D- (including resistors and capacitor)
https://camo.githubusercontent.com/f6eeaf4a553951599a5c2197d05be781ac764dd330e77d7f09aeae72b76785dd/68747470733a2f2f6465767a6f6e652e6e6f7264696373656d692e636f6d2f726573697a65642d696d6167652f5f5f73697a652f373032783437362f5f5f6b65792f737570706f72742d6174746163686d656e74732f62656566356431623737363434633434386461626666333136363866336134372d32643162336461623134623834646561613262313637646539626261363762612f5343482e504e47

When i connect the USB cable, nothing shows up, not in Arduino IDE and not in finder (i'm using mac).
Is my way of doing this correct or did i do something wrong?

danielkucera, 2023-02-20:
It expects to get .bin file, not .hex.
There is a WiP PR for hex file support #15

bbuster2, 2023-02-20:
Well, i tried the hex file first but then converted it to .bin via HxD but this was the result from the bin upload

danielkucera, 2023-02-20:
Did you erase it after each flash?

bbuster2, 2023-02-20:
Yes, i just tried it again, connect the ESP32, erased the chip and flashed the bootloader file (hex file converted to bin file).

Is the output i showed above correct?

Bluebird-Software, 2023-02-20:
Update,

I tried to install arduino_nano_33_ble_bootloader-0.7.0_s140_6.1.1.bin
When i flashed it via the ESP32, the arduino IDE showed except my ESP32 port, also a new port: /dev/cu.usbmodemF2BD185AD80C1

When i unplug the ESP32, the port goes away.
If i keep the ESP32 plugged in and try to upload a new arduino program to the new port, it cannot upload

>Sketch uses 138160 bytes (16%) of program storage space. Maximum is 815104 bytes.
Global variables use 13836 bytes (5%) of dynamic memory, leaving 223732 bytes for local variables. Maximum is 237568 bytes.
Performing 1200-bps touch reset on serial port /dev/cu.usbmodemF2BD185AD80C1
Waiting for upload port...
No upload port found, using address:"/dev/cu.usbmodemF2BD185AD80C1"  label:"/dev/cu.usbmodemF2BD185AD80C1"  protocol:"serial"  protocol_label:"Serial Port (USB)"  properties:{key:"pid"  value:"0x521F"}  properties:{key:"serialNumber"  value:"F2BD185AD80C"}  properties:{key:"vid"  value:"0x1915"} as fallback`

bbuster2, 2023-02-20:
Update,

I tried to install arduino_nano_33_ble_bootloader-0.7.0_s140_6.1.1.bin
When i flashed it via the ESP32, the arduino IDE showed except my ESP32 port, also a new port: /dev/cu.usbmodemF2BD185AD80C1

When i unplug the ESP32, the port goes away.
If i keep the ESP32 plugged in and try to upload a new arduino program to the new port, it cannot upload

Sketch uses 138160 bytes (16%) of program storage space. Maximum is 815104 bytes.
Global variables use 13836 bytes (5%) of dynamic memory, leaving 223732 bytes for local variables. Maximum is 237568 bytes.
Performing 1200-bps touch reset on serial port /dev/cu.usbmodemF2BD185AD80C1
Waiting for upload port...
No upload port found, using address:"/dev/cu.usbmodemF2BD185AD80C1" label:"/dev/cu.usbmodemF2BD185AD80C1" protocol:"serial" protocol_label:"Serial Port (USB)" properties:{key:"pid" value:"0x521F"} properties:{key:"serialNumber" value:"F2BD185AD80C"} properties:{key:"vid" value:"0x1915"} as fallback`
Files are downloaded from here https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/tag/0.7.0
The zip files include a .bin file

bbuster2, 2023-02-20:
It looks like flashing is not doing anything.
After i flashed the file and read the nrf register, all the values are (tested with value 0, 300 and 10000):
```
Ok: Register read address: 0x00000000 value: 0x00000000
Ok: Register read address: 0x00000300 value: 0x00000000
Ok: Register read address: 0x00010000 value: 0x00000000
...
```

bbuster2, 2023-02-21:
After some more tests i see different bootloader files get different values than 0x0000000
But i still cannot upload my own code to the usb device, also when unplugging the ESP32, the usbmodem device goes away as well

danielkucera, 2023-02-21:
update to latest master (a2d9a3d) and try that hex file again

danielkucera, 2023-02-21:
What does the serial log say?

bbuster2, 2023-02-21:
From Arduino IDE -> Serial monitor while "flashing" the firmware?

danielkucera, 2023-02-21:
the same way you got this:
```
....
14:55:22.959 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:22.992 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:22.992 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.025 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.025 -> 000 Write Register: 0x4001e504 : 0x00000001
14:55:23.025 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.057 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.057 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.090 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.090 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.121 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.121 -> 0000 Read Register: 0x4001e400 : 0x00000000
14:55:23.121 -> Done flashing file, it took 19015ms speed: 28.0205kbs
```

bbuster2, 2023-02-21:
```
13:27:37.024 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.057 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.057 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.057 -> 000 Write Register: 0x4001e504 : 0x00000001
13:27:37.089 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.089 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.121 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.121 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.154 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.154 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.188 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.188 -> 000 Write Register: 0x4001e504 : 0x00000001
13:27:37.219 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.219 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.251 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.251 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.285 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.285 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.317 -> 0000 Read Register: 0x4001e400 : 0x00000000
13:27:37.317 -> Done flashing file, it took 16648ms speed: 32.0044kbs
```

You need all? Because i cannot copy everything the Serial monitor outputs so if you need all i need to use a different way to copy the result

bbuster2, 2023-02-22:
Does this give any information?

danielkucera, 2023-02-23:
No, you need to see something like: `hex line found:`.
But I just noticed it only works with Flash uploaded file:
[image]

Bluebird-Software, 2023-02-25:
Thank you for your reply.

I don't get it working so here are my steps / results:

First i turn on the ESP32 with the nRF52840 connected.

1. Open swd.local
2. Click Init SWD -> Ok: Init of SWD ID: 0x2ba01477
3. Erase nRF -> Ok: Everything erased
>0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000

4. Flash Uploaded File -> Select file (Seeed_XIAO_nRF52840_Sense_bootloader-0.6.1_s140_7.3.0), offset = 0
5. Flash uploaded File

>received data file:Seeed_XIAO_nRF52840_Sense_bootloader-0.6.1_s140_7.3.0.hex index:0 len:1116 final:0
hex line found: 
:04000003F000AE1942
hex line found: 
:020000040000FA
Setting flash_offset to 0
hex line found: 
:1000000000040020810A000015070000610A0000BA
Writing flash_offset: 0x0 off 0x0
000 Write Register: 0x4001e504 : 0x00000001
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
hex line found: 
:100010001F07000029070000330700000000000050
Writing flash_offset: 0x0 off 0x10
000 Write Register: 0x4001e504 : 0x00000001
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
hex line found: 
:10002000000000000000000000000000A50A000021
..

Now i have a different problem, after half a minute the ESP32 crashes and gives me this error, while i am almost certain that it completed the flash before. It does it now on multiple .hex files, even after reuploading the firmware to the ESP32

>hex line found: 
:10056000704770B50C4605464FF4806608E0284693
Writing flash_offset: 0x0 off 0x560
000 Write Register: 0x4001e504 : 0x00000001
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
E (97281) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (97281) task_wdt:  - async_tcp (CPU 1)
E (97281) task_wdt: Tasks currently running:
E (97281) task_wdt: CPU 0: IDLE0
E (97281) task_wdt: CPU 1: loopTask
E (97281) task_wdt: Aborting.
abort() was called at PC 0x40159604 on core 0
ELF file SHA256: 0000000000000000
Backtrace: 0x4008cbec:0x3ffbf850 0x4008ce65:0x3ffbf870 0x40159604:0x3ffbf890 0x40089655:0x3ffbf8b0 0x40172327:0x3ffbc220 0x4015afab:0x3ffbc240 0x4008f615:0x3ffbc260 0x4008de6a:0x3ffbc280
Rebooting...

bbuster2, 2023-02-25:
Thank you for your reply.

I don't get it working so here are my steps / results:

First i turn on the ESP32 with the nRF52840 connected.

1. Open swd.local
2. Click Init SWD -> Ok: Init of SWD ID: 0x2ba01477
3. Erase nRF -> Ok: Everything erased
>0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000

4. Flash Uploaded File -> Select file (Seeed_XIAO_nRF52840_Sense_bootloader-0.6.1_s140_7.3.0), offset = 0
5. Flash uploaded File

>received data file:Seeed_XIAO_nRF52840_Sense_bootloader-0.6.1_s140_7.3.0.hex index:0 len:1116 final:0
hex line found: 
:04000003F000AE1942
hex line found: 
:020000040000FA
Setting flash_offset to 0
hex line found: 
:1000000000040020810A000015070000610A0000BA
Writing flash_offset: 0x0 off 0x0
000 Write Register: 0x4001e504 : 0x00000001
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
hex line found: 
:100010001F07000029070000330700000000000050
Writing flash_offset: 0x0 off 0x10
000 Write Register: 0x4001e504 : 0x00000001
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
hex line found: 
:10002000000000000000000000000000A50A000021
..

Now i have a different problem, after half a minute the ESP32 crashes and gives me this error, while i am almost certain that it completed the flash before. It does it now on multiple .hex files, even after reuploading the firmware to the ESP32

>hex line found: 
:10056000704770B50C4605464FF4806608E0284693
Writing flash_offset: 0x0 off 0x560
000 Write Register: 0x4001e504 : 0x00000001
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
0000 Read Register: 0x4001e400 : 0x00000000
E (97281) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (97281) task_wdt:  - async_tcp (CPU 1)
E (97281) task_wdt: Tasks currently running:
E (97281) task_wdt: CPU 0: IDLE0
E (97281) task_wdt: CPU 1: loopTask
E (97281) task_wdt: Aborting.
abort() was called at PC 0x40159604 on core 0
ELF file SHA256: 0000000000000000
Backtrace: 0x4008cbec:0x3ffbf850 0x4008ce65:0x3ffbf870 0x40159604:0x3ffbf890 0x40089655:0x3ffbf8b0 0x40172327:0x3ffbc220 0x4015afab:0x3ffbc240 0x4008f615:0x3ffbc260 0x4008de6a:0x3ffbc280
Rebooting...

bbuster2, 2023-02-25:
Update:

I got a bit further. After selecting another Arduino bootloader file (pca10056_bootloader-0.3.0_s140_6.1.1.hex), a new USB device got detected (NRF52BOOT).

I could now upload an example sketch via Arduino IDE and i got this:

```
Opened serial port /dev/cu.usbmodem1301
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 113640
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
########################################
########################################
########################################
######################
Activating new firmware

DFU upgrade took 7.148498058319092s
Device programmed.
```

After that, nothing happens (except Mac says the USB device is disconnected).
After double resetting the device, the NRF52BOOT shows up again and i see these 3 files:

- CURRENT.UF2
- INDEX.HTM
- INFO_UF2.TXT

But the serial monitor doesn't show any data, and the nrf beacon app (for ios) says the beacon is not found.
What am i doing wrong?
And just powering the nrf up isn't enough to make it start the normal way? I need to double reset it to get the NRF52BOOT usb device come up