atc1441/ESP32_nRF52_SWD #32: Following Readme doesn't compile. Do this steps TO FIX in ARDUINO

JKSniper1, 2024-09-12:
Following each step exactly, with latest Arduino IDE 2.3.2. Install ESP32 boad version 1.0.6. With latest, it won't work, don't even try. I've wasted like few hours before figuring out what was the problem and how to fix it. I was trying both in Arduino and Platformio. In the end Arduino worked. 

To make it work: 

Delete (or move somewhere else) all other SIMILAR libraries that you might have, as they might have same names and would conflict hard with these! 

When you have new libraries inside arduino library folder and you try to compile, it will show: 

1st problem. No secrets.h file (to fix, generate or make one with .py, then include in arduino by #include "secrets.h") You can also make new file in Arduino and just add this line into it: #define AP_PASSWORD "password"  , where "password" is the pass you need to input when trying to connect to AutoConnectAP SSID. Why this is so complicated, I don't know :) 

2nd problem. no defined pins (to fix, go to defines.h tab and uncomment LED, GLITHCHER.... etc.) and comment or delete first line:
 // move to defines in platformio.ini

3th problem. SPIFFSEditor.cpp and SPIFFSEditor.h are overwriting as they are defined (probably) in library aswell. To fix this, go in Arduino tabs and delete this two tabs. 

Now it should hopefully compile and you can compile and upload, and follow the rest of the Readme steps. 

For author: Please update ReadMe to be either for Platformio or Arduino. Not mixed, as nothing works now.

Kzen-lab, 2024-09-29:
Hello, I have the same problems, errors occur while compiling the latest code for Arduino. Could you add a read me edit please?

atc1441, 2025-02-27:
Updated to the latest working versions and switched to PlatformIO only support :)

closed 2025-02-27 by atc1441