atc1441/ESP32_nRF52_SWD #36: Unable to flash nrf52840 nice!nano

wesselah, 2025-08-08:
I successfully compiled and installed the software on an esp32 wroom and when i upload the html file the webserver dashboard showed. When i connect the nice!nano with the pads on the back i can "as it seems" init sw and al the parameters of the board are shown. I can erase and i can upload but whatever i do no bootloader gets installed . It is not clear to me if i  should use  hex or bin files  and where to put them selecting a file and then select flash turns in a endless loop on the webpage. When i upload a file to the edit page and then select it in the webpage, it seems to sart uploadingand it says it is finised  but never a disk shows when hitting reset twice any explaination would be greatly appreciated

wesselah, 2025-08-08:
Just to give some background a com port never shows up from the nrf52840 when i plugin usb

valzzu, 2025-09-27:
> I successfully compiled and installed the software on an esp32 wroom and when i upload the html file the webserver dashboard showed. When i connect the nice!nano with the pads on the back i can "as [...]

flashing the bootloader isnt fast, the infinite spinning means its flashing, just give it time.